litex-hub/linux-on-litex-vexriscv #383: After booting to [10.122727] Waiting for root device /dev/mmcblk0p2... there is no response.

hex55, 2024-05-15:
After booting to [10.122727] Waiting for root device /dev/mmcblk0p2... there is no response. The rootfs is in the SD card and is in ext4 format. Do you have any reference? Is there any reference material on booting from SD card? Thanks!
[image]

Dolu1990, 2024-05-15:
When you boot using the ramfs, can you get the sdcard to work in the linux terminal ? that would be the first easy step (if you can boot with the ramfs ?)

Dolu1990, 2024-05-15:
(I do not have any self contained reference)

hex55, 2024-05-16:
> When you boot using the ramfs, can you get the sdcard to work in the linux terminal ? that would be the first easy step (if you can boot with the ramfs ?)

Thank you, My rom cannot run rootfs because it is only 8M. This morning I successfully ran linux on the sdcard, yes, it can only be used as a toy.

hex55, 2024-05-16:
> (I do not have any self contained reference)

ok. I still thank you!

closed 2024-05-16 by hex55