litex-hub/linux-on-litex-vexriscv #386: running on hardware

azaad-deepgrid, 2024-05-29:
[image]
i connected the board and I got this issue after running ./make.py --board=arty --variant=a7_100 --cpu_count=1 --build .
 In --init__.py there is nothing written but i could not exactly find the issue and resolve it.

trabucayre, 2024-05-29:
Are you sure to have a linux-on-linux-vexriscv up to date ? 
For the same line I see this [import](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L76).

trabucayre, 2024-05-29:
Weird! according to [this line](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L77)
its:
```python
Board.__init__(self, digilent_arty.BaseSoC, soc_capabilities={
```
and not
```python
Board.__init__(self, arty.BaseSoC, soc_capabilities={
```

Look like you have an issue with your local version

azaad-deepgrid, 2024-05-30:
Can you provide me the way where i can update everything in my local.

trabucayre, 2024-05-31:
with `git pull` ?