litex-hub/linux-on-litex-vexriscv #387: extension zicsr is missing

LaneaLucy, 2024-05-29:
hey, i tried to compile linux but i get a shit load of the same error, extension zicsr is missing. anyone know how to fix?

here a example of the error message
```
/home/lanealucy/litex/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_ecall_hsm.c:25: Error: unrecognized opcode `csrr s4,0x340', extension `zicsr' required
```
the only change i made from documentation to get this far is to let buildroot set gcc version higher because other version is depracted

trabucayre, 2024-05-29:
Could you try to comment [these 4 lines](https://github.com/litex-hub/opensbi/blob/1.3.1-linux-on-litex-vexriscv/platform/litex/vexriscv/objects.mk#L20-L23)? without `PLATFORM_RISCV_xxx` opensbi will detect config by itself.

LaneaLucy, 2024-05-29:
@trabucayre now i get the following errors...

```
>>> opensbi 1.3.1-linux-on-litex-vexriscv Building
GIT_DIR=. PATH="/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/bin:/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/sbin:/usr/lib/ccache:/home/lanealucy/Gowin_V1.9.9.03_Education_linux/IDE/bin:/home/lanealucy/.local/bin:/home/lanealucy/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" CROSS_COMPILE=/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/bin/riscv32-buildroot-linux-gnu-  PLATFORM=litex/vexriscv /usr/bin/make -j5 -C /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv
 ELF       platform/litex/vexriscv/firmware/fw_dynamic.elf
 ELF       platform/litex/vexriscv/firmware/fw_payload.elf
 ELF       platform/litex/vexriscv/firmware/fw_jump.elf
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/lib/libplatsbi.a(sbi_hart.o): in function `mstatus_init':
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_hart.c:109:(.text+0x51c): undefined reference to `put_f32_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/lib/libplatsbi.a(sbi_misaligned_ldst.o): in function `sbi_misaligned_load_handler':
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:157:(.text+0x1e4): undefined reference to `put_f32_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:114:(.text+0x210): undefined reference to `put_f64_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/lib/libplatsbi.a(sbi_misaligned_ldst.o): in function `sbi_misaligned_store_handler':
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:222:(.text+0x33c): undefined reference to `get_f64_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:259:(.text+0x/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/lib/libplatsbi.a(sbi_hart.o)3f0): undefined reference to `: in function `mstatus_init':
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_hart.c:109:(.text+0x51c): undefined reference to `put_f32_reg'
get_f64_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:232:(.text+0x438): undefined reference to `get_f32_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:187:(.text+0x47e): undefined reference to `get_f64_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/sbi_misaligned_ldst.c:249:(.text+0x4e0): undefined reference to `get_f32_reg'
/home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/host/lib/gcc/riscv32-buildroot-linux-gnu/13.2.0/../../../../riscv32-buildroot-linux-gnu/bin/ld.bfd: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party [... cut ...]
make[2]: *** [Makefile:555: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/firmware/fw_payload.elf] Fehler 1
collect2: Fehler: ld gab 1 als Ende-Status zurück
make[2]: *** [Makefile:555: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/build/platform/litex/vexriscv/firmware/fw_dynamic.elf] Fehler 1
make[1]: *** [package/pkg-generic.mk:283: /home/lanealucy/litex/linux-on-litex-vexriscv/third_party/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/.stamp_built] Fehler 2
make: *** [Makefile:82: _all] Fehler 2
```

LaneaLucy, 2024-05-30:
ok, now i did a distclean and it build properly. its uploading right now so wish me luck XD

LaneaLucy, 2024-05-30:
cool, now i get crc error when trying to upload...
should my Image really be 8,9M big?

LaneaLucy, 2024-05-30:
Tried to compile it again fresh, still 8,9M

LaneaLucy, 2024-05-30:
tried the prebuild but nothing happens...

[image]

watermeko, 2024-10-11:
@LaneaLucy Try to add arguments. `--cpu-type vexriscv_smp --cpu-variant linux`