litex-hub/linux-on-litex-vexriscv #389: ResolutionError$CantDownloadModule: Error downloading org.scala-sbt:sbt:1.6.0

jjagielo, 2024-06-04:
I am currently running sbt 1.10.0 on my Ubuntu VM. When trying to run
```bash
./sim.py
```
I am met with the following error:
```bash
[info] [launcher] getting org.scala-sbt sbt 1.6.0  (this may take some time)...
[error] [launcher] xsbt.boot.internal.shaded.coursier.error.ResolutionError$CantDownloadModule: Error downloading org.scala-sbt:sbt:1.6.0
  not found: /home/jason/.ivy2/local/org.scala-sbt/sbt/1.6.0/ivys/ivy.xml
  download error: Caught java.net.ConnectException (Connection refused (Connection refused)) while downloading https://repo1.maven.org/maven2/org/scala-sbt/sbt/1.6.0/sbt-1.6.0.pom
  download error: Caught java.net.ConnectException (Connection refused (Connection refused)) while downloading https://repo.scala-sbt.org/scalasbt/maven-releases/org/scala-sbt/sbt/1.6.0/sbt-1.6.0.pom
  download error: Caught java.net.ConnectException (Connection refused (Connection refused)) while downloading https://repo.scala-sbt.org/scalasbt/maven-snapshots/org/scala-sbt/sbt/1.6.0/sbt-1.6.0.pom
  download error: Caught java.net.ConnectException (Connection refused (Connection refused)) while downloading https://repo.typesafe.com/typesafe/ivy-releases/org.scala-sbt/sbt/1.6.0/ivys/ivy.xml
  download error: Caught java.net.ConnectException (Connection refused (Connection refused)) while downloading https://repo.scala-sbt.org/scalasbt/ivy-snapshots/org.scala-sbt/sbt/1.6.0/ivys/ivy.xml
[error] [launcher] could not retrieve sbt 1.6.0
Traceback (most recent call last):
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./sim.py", line 177, in <module>
    main()
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./sim.py", line 168, in main
    builder.build(sim_config=sim_config,
  File "/usr/local/lib/python3.12/dist-packages/litex-2023.12-py3.12.egg/litex/soc/integration/builder.py", line 372, in build
    self.soc.finalize()
  File "/usr/local/lib/python3.12/dist-packages/litex-2023.12-py3.12.egg/litex/soc/integration/soc.py", line 1442, in finalize
    Module.finalize(self)
  File "/usr/local/lib/python3.12/dist-packages/migen-0.9.2-py3.12.egg/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/migen-0.9.2-py3.12.egg/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
                    ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/dist-packages/migen-0.9.2-py3.12.egg/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/usr/local/lib/python3.12/dist-packages/migen-0.9.2-py3.12.egg/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/usr/local/lib/python3.12/dist-packages/litex-2023.12-py3.12.egg/litex/soc/cores/cpu/vexriscv_smp/core.py", line 578, in do_finalize
    self.add_sources(self.platform)
  File "/usr/local/lib/python3.12/dist-packages/litex-2023.12-py3.12.egg/litex/soc/cores/cpu/vexriscv_smp/core.py", line 428, in add_sources
    self.generate_netlist()
  File "/usr/local/lib/python3.12/dist-packages/litex-2023.12-py3.12.egg/litex/soc/cores/cpu/vexriscv_smp/core.py", line 310, in generate_netlist
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python3.12/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv && sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --reset-vector=0 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=32 --aes-instruction=False --expose-time=False --out-of-order-decoder=True --privileged-debug=False --hardware-breakpoints=1 --wishbone-memory=False --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood_Hb1 --netlist-directory=/usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4 --jtag-tap=False"' returned non-zero exit status 1
```
Is this related to my installation of sbt? I've tried looking to other places and have not found anyone with a similar error. Also, when going to some of the links that caused the java.net.ConnectException, some seemed to not exist (Either pinging or searching them manually), such as [here](https://scala.jfrog.io/ui/native/ivy-releases/org.scala-sbt/sbt/1.6.0/ivys/ivy.xml):
```json
{
  "errors": [
    {
      "status": 404,
      "message": "Item ivy-releases:org.scala-sbt/sbt/1.6.0/ivys does not exist"
    }
  ]
}
```

Dolu1990, 2024-06-04:
Hi,

It should be now fixed with : 
https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/commit/1f8f8bb49538663759f043e83ebae311132fc3e5

jjagielo, 2024-06-04:
It is still trying to get version 1.6.0 of sbt, unsure if that's just because of a print statement since I saw that the PR looked to update the _build.properties_ to sbt 1.10.0. Also still leads to the 1.6.0 ivy.xml files that give the same .xml file as before.
[image]

Dolu1990, 2024-06-05:
Hmm weird.
Did you checked that your local copy of pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project/build.properties has sbt 1.10.0 set ?

For me it uses 1.10.0 when i run litex.

jjagielo, 2024-06-05:
[image]
It's set to 1.10.0 so that's not the issue. I looked around at other _build.properties_ configurations to see if they were still set to 1.6.0 and the only discrepency was here, but that was with sbt 1.9.7, not 1.6.0 so that wouldn't be the issue.
[image]

Dolu1990, 2024-06-05:
linux-on-litex-vexriscv use pythondata-cpu-vexriscv-smp, not pythondata-cpu-vexriscv

The issue you have is kinda weird, i'm not aware of any other place where the sbt version is specified.
Hmmm, maybe try to delete /home/jason/.ivy2 , aswell as deleting the whole pythondata-cpu-vexriscv-smp and recloning it (to be sure there is no dirty cache in it)

let's me know if it arrenge things, else, can you send me the full litex stdout in your terminal ?

jjagielo, 2024-06-05:
Deleted /home/jason/.ivy2 as well as reverified my installation of sbt. It's all working now, thank you for your help!

closed 2024-06-05 by jjagielo

Dolu1990, 2024-06-05:
> Deleted /home/jason/.ivy2 as well as reverified my installation of sbt. It's all working

Nice :D (i was about to be hopeless XD)