litex-hub/linux-on-litex-vexriscv #390: SpinalHDL git repository not detected

jjagielo, 2024-06-05:
Running 
```bash
./sim.py
```
and I'm currently getting this error:
```bash
INFO:SoC:--------------------------------------------------------------------------------
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw32_Ood_Hb1
Generating cluster netlist
[info] welcome to sbt 1.10.0 (Temurin Java 1.8.0_412)
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/jason/.sbt/1.0/staging/7a0d3e6968cc5c46ea70/spinalhdl/project
[info] loading settings for project all from build.sbt ...
[info] set current project to VexRiscv (in build file:/usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
fatal: not a git repository: /home/jason/.sbt/1.0/staging/7a0d3e6968cc5c46ea70/spinalhdl/../../../../.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL
[error] java.lang.RuntimeException: Nonzero exit value: 128
[error]         at scala.sys.package$.error(package.scala:30)
[error]         at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.slurp(ProcessBuilderImpl.scala:138)
[error]         at scala.sys.process.ProcessBuilderImpl$AbstractBuilder.$bang$bang(ProcessBuilderImpl.scala:108)
[error]         at $5335fef8e99ff34f0fec$.liftedTree1$1(build.sbt:87)
[error]         at $5335fef8e99ff34f0fec$.gitHash(build.sbt:86)
[error]         at $5335fef8e99ff34f0fec$.$anonfun$core$8(build.sbt:151)
[error]         at scala.Function1.$anonfun$compose$1(Function1.scala:49)
[error]         at sbt.internal.util.$tilde$greater.$anonfun$$u2219$1(TypeFunctions.scala:63)
[error]         at sbt.std.Transform$$anon$4.work(Transform.scala:69)
[error]         at sbt.Execute.$anonfun$submit$2(Execute.scala:283)
[error]         at sbt.internal.util.ErrorHandling$.wideConvert(ErrorHandling.scala:24)
[error]         at sbt.Execute.work(Execute.scala:292)
[error]         at sbt.Execute.$anonfun$submit$1(Execute.scala:283)
[error]         at sbt.ConcurrentRestrictions$$anon$4.$anonfun$submitValid$1(ConcurrentRestrictions.scala:265)
[error]         at sbt.CompletionService$$anon$2.call(CompletionService.scala:65)
[error]         at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error]         at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
[error]         at java.util.concurrent.FutureTask.run(FutureTask.java:266)
[error]         at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
[error]         at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
[error]         at java.lang.Thread.run(Thread.java:750)
[error] (ProjectRef(uri("file:/usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/"), "core") / Compile / managedSources) Nonzero exit value: 128
```
I'm aware that it says that the given directory does not exist, however, when I run the following, it says the HEAD is detached:
```bash
jason@jason-VirtualBox:~/Desktop/LiteX/pythondata-cpu-vexriscv-smp/.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL$ git status
HEAD detached at 002ffd50b
nothing to commit, working tree clean
```
Would it be safe to just manually clone the SpinalHDL git repository, or is there a proper way to ensure that this can detect it as the proper git repository?

Dolu1990, 2024-06-05:
Hi,
Ahh that may be due to some of the maniuplation done on the other issue we just resolved.
When you git clone the pythondata-vexriscv you need to do it with the  --recursive argument.
Could it be the reason ?

jjagielo, 2024-06-05:
Which exact directory needs to be cloned using the --recursive argument?
[image]
I'll give it a shot and see if the issue is resolved by recloning it.

jjagielo, 2024-06-05:
I assume it's this one? --> https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/tree/master

Dolu1990, 2024-06-06:
Hmm this is weird
Do you have the same git hash for SpinalHDL and VexRiscv than specified here : 
https://github.com/litex-hub/pythondata-cpu-vexriscv_smp/tree/master/pythondata_cpu_vexriscv_smp/verilog/ext 
?

jjagielo, 2024-06-06:
I cloned https://github.com/SpinalHDL/SpinalHDL/tree/002ffd50bca59e0f8b9c74b0d00cac86d8379810 and got a different hash than the one above. Should I just manually clone it where the incorrect hash is?

Dolu1990, 2024-06-07:
> Should I just manually clone it where the incorrect hash is?

Yes, you realy need to have the correct git commit hash

jjagielo, 2024-06-10:
Just out of curiosity, the path specified in the error is here: 
```bash
fatal: not a git repository: /home/jason/.sbt/1.0/staging/7a0d3e6968cc5c46ea70/spinalhdl/../../../../.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL
```
which would be the same path as...
```bash
/home/jason/.sbt/.git/modules/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL
```
This path does not exist. Is this the path I need to reclone to repository for SpinalHDL (_/LiteX_ holds all the files for this project)?
```bash
/home/jason/Desktop/LiteX/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL
```

Dolu1990, 2024-06-10:
that is realy weird XD

Can you try to :

```shell
cd /home/jason/Desktop/LiteX
rm -rf  pythondata-cpu-vexriscv-smp
git clone https://github.com/litex-hub/pythondata-cpu-vexriscv_smp.git --recursive
```

And then retry the generation ?

Just to be sure we are in sync

jjagielo, 2024-06-10:
I ran everything as you said, and it's still throwing the same error with the exact same path. 
```bash
jason@jason-VirtualBox:~/Desktop/LiteX$ git clone https://github.com/litex-hub/pythondata-cpu-vexriscv_smp.git --recursive
Cloning into 'pythondata-cpu-vexriscv_smp'...
remote: Enumerating objects: 348, done.
remote: Counting objects: 100% (259/259), done.
remote: Compressing objects: 100% (50/50), done.
remote: Total 348 (delta 225), reused 234 (delta 208), pack-reused 89
Receiving objects: 100% (348/348), 4.54 MiB | 1.75 MiB/s, done.
Resolving deltas: 100% (230/230), done.
Submodule 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL' (https://github.com/SpinalHDL/SpinalHDL.git) registered for path 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL'
Submodule 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv' (https://github.com/SpinalHDL/VexRiscv.git) registered for path 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv'
Cloning into '/home/jason/Desktop/LiteX/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL'...
remote: Enumerating objects: 109411, done.
remote: Counting objects: 100% (8574/8574), done.
remote: Compressing objects: 100% (4116/4116), done.
remote: Total 109411 (delta 5124), reused 6559 (delta 3289), pack-reused 100837
Receiving objects: 100% (109411/109411), 66.36 MiB | 1.83 MiB/s, done.
Resolving deltas: 100% (52711/52711), done.
Cloning into '/home/jason/Desktop/LiteX/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv'...
remote: Enumerating objects: 16588, done.
remote: Counting objects: 100% (4444/4444), done.
remote: Compressing objects: 100% (1073/1073), done.
remote: Total 16588 (delta 3509), reused 4088 (delta 3348), pack-reused 12144
Receiving objects: 100% (16588/16588), 13.06 MiB | 1.72 MiB/s, done.
Resolving deltas: 100% (9973/9973), done.
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL': checked out '002ffd50bca59e0f8b9c74b0d00cac86d8379810'
Submodule 'tester/src/test/python/cocotblib' (https://github.com/SpinalHDL/CocotbLib.git) registered for path 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/tester/src/test/python/cocotblib'
Cloning into '/home/jason/Desktop/LiteX/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/tester/src/test/python/cocotblib'...
remote: Enumerating objects: 124, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 124 (delta 45), reused 43 (delta 43), pack-reused 68
Receiving objects: 100% (124/124), 30.72 KiB | 241.00 KiB/s, done.
Resolving deltas: 100% (68/68), done.
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/tester/src/test/python/cocotblib': checked out 'a98830423924fc89bfebae84cb802fc90d352602'
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv': checked out '625fc8437952ca00d474f0547faef72cbe68a32c'
Submodule 'src/test/resources/VexRiscvRegressionData' (https://github.com/SpinalHDL/VexRiscvRegressionData.git) registered for path 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/src/test/resources/VexRiscvRegressionData'
Cloning into '/home/jason/Desktop/LiteX/pythondata-cpu-vexriscv_smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/src/test/resources/VexRiscvRegressionData'...
remote: Enumerating objects: 78, done.
remote: Total 78 (delta 0), reused 0 (delta 0), pack-reused 78
Receiving objects: 100% (78/78), 10.88 MiB | 1.76 MiB/s, done.
Resolving deltas: 100% (2/2), done.
Submodule path 'pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/src/test/resources/VexRiscvRegressionData': checked out '539398c1481203a51115b5f1228ea961f0ac9bd3'
```
Above is the output when cloning the repo, in-case there is anything interesting that might help you find the issue.

Dolu1990, 2024-06-10:
Thanks
So, this week i just reinstalled my laptop from scratch, and so with fresh litex, running for instance: 
litex_sim --cpu-type=vexriscv_smp --cpu-count=3

I think your install  of litex isn't done via the https://github.com/enjoy-digital/litex?tab=readme-ov-file#quick-start-guide, as i can now see your : 
[info] loading project definition from /usr/local/lib/python3.12/dist-packages/pythondata_cpu_vexriscv_smp-1.0.1.post325-py3.12.egg/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project

How did you installed litex ?

jjagielo, 2024-06-10:
I just followed here, https://github.com/litex-hub/linux-on-litex-vexriscv/?tab=readme-ov-file#-installing-litex, but I'll follow the link you just sent and see if that addresses the issue

jjagielo, 2024-06-10:
I'm running the _./litex_setup.py_ in _./Litex/linux-on-litex-vexriscv_

Dolu1990, 2024-06-10:
hmm, it seems that you may have a broken litex intall.
If i remember well, there is some arguements to the litex_setup.py, maybe there is one to uninstall ?

closed 2024-06-10 by jjagielo