litex-hub/linux-on-litex-vexriscv #391: Error: unrecognized opcode `csrr s5,0xf14', extension `zicsr' required

phillippZZ, 2024-06-05:
Hi, everyone.

I tried to build a linux image with following commands:
> $ git clone http://github.com/buildroot/buildroot
$ cd buildroot
$ make BR2_EXTERNAL=../linux-on-litex-vexriscv/buildroot/ litex_vexriscv_defconfig
$ make menuconfig (And I checked out "gcc 11.x support removed" in the legacy config options)
$ make

Then I got a lot of errors due to "ziscr":
> /usr/src/buildroot/output/build/opensbi-1.3.1-linux-on-litex-vexriscv/lib/sbi/riscv_asm.c:195: Error: unrecognized opcode `csrw 0x3b0+32+16+0+0+2+1,a1', extension `zicsr' required

How can I solve this issue?

Dolu1990, 2024-06-05:
hmm maybe replacing https://github.com/litex-hub/linux-on-litex-vexriscv/blob/ea17e31537cdabe780cf5747cbe5b37ea09f717f/buildroot/configs/litex_vexriscv_defconfig#L25C1-L25C21 with BR2_GCC_VERSION_12_X=y ?

Not sure

phillippZZ, 2024-06-05:
It doesn't work. And I tried to add "March=rv32i_zicsr" somewhere for gcc but couldn't find a place to add it😅

Dolu1990, 2024-06-10:
This is related to : 

# Bootloader (opensbi)
BR2_TARGET_OPENSBI=y
BR2_TARGET_OPENSBI_CUSTOM_GIT=y
BR2_TARGET_OPENSBI_CUSTOM_REPO_URL="https://github.com/litex-hub/opensbi.git"
BR2_TARGET_OPENSBI_CUSTOM_REPO_VERSION="1.3.1-linux-on-litex-vexriscv"
BR2_TARGET_OPENSBI_PLAT="litex/vexriscv"
BR2_TARGET_OPENSBI_INSTALL_DYNAMIC_IMG=n

That opensbi branch seems to bo too outdated.

trabucayre, 2024-06-10:
@Dolu1990 this branch is used in [CI](https://github.com/enjoy-digital/litex_hw_ci/) with mainly one [patch](https://github.com/enjoy-digital/litex_hw_ci/blob/main/linux/buildroot/patches/opensbi/1.3.1-linux-on-litex-vexriscv/0003-architecture_autodetect.patch)
I have observed gateware and buildroot MUST be coherent (RVC is enabled in config, `--with-rvc` must be added)

phillippZZ, 2024-06-10:
Sorry, I don't know what I'm supposed to do now. Do I use "https://github.com/riscv-software-src/opensbi" in "litex_vexriscv_defconfig", or keep using "https://github.com/litex-hub/opensbi.git" and follow @trabucayre 's instruction?

Dolu1990, 2024-06-10:
@trabucayre My bad, you are right ^^ the version is good.

@phillippZZ 
By a happy hazard, i just hit the same issue.

I got things to compile with  (didn't used RVC): 

```shell
make   ... \
     PLATFORM_RISCV_XLEN=32 \
     PLATFORM_RISCV_ISA=rv32ima_zicsr_zifencei \
     PLATFORM_RISCV_ABI=ilp32 \
     ...
```

phillippZZ, 2024-06-10:
@Dolu1990 Thank you.

Do I download opensbi 1.4 and compile it with something like:
> $ wget https://github.com/riscv-software-src/opensbi/archive/refs/tags/v1.4.zip
$ unzip v.1.4.zip
$ cd opensbi
$ make CROSS_COMPILE=riscv-none-embed- PLATFORM=litex/vexriscv PLATFORM_RISCV_XLEN=32 PLATFORM_RISCV_ISA=rv32ima_zicsr_zifencei PLATFORM_RISCV_ABI=ilp32

Also, should I modify litex_vexriscv_defconfig as the following:
> BR2_TARGET_OPENSBI=y
BR2_TARGET_OPENSBI_CUSTOM_GIT=n
BR2_TARGET_OPENSBI_CUSTOM_DIR="/path/to/local/opensbi"
BR2_TARGET_OPENSBI_PLAT="litex/vexriscv"
BR2_TARGET_OPENSBI_INSTALL_DYNAMIC_IMG=n

Dolu1990, 2024-06-13:
Don't use  opensbi 1.4, as i proposed, it is for when you use the generic opensbi platform, which is the same as proposed on linux-on-litex-vexriscv.

How did you made your program file ?

phillippZZ, 2024-06-13:
I didn't use opensbi 1.4 or build the linux image. I simply compiled a project called IsolBench with "riscv64-unknown-linux-gnu-gcc" and moved it into the rootf.cpio.
[image]
As you can see, some of the scripts don't work since the linux image doesn't support certain functionalities.

Could you tell me how I can solve these issues?  And IsolBench can be downloaded from "https://github.com/CSL-KU/IsolBench"

closed 2024-07-21 by phillippZZ