litex-hub/linux-on-litex-vexriscv #392: linux doesnt boot

LaneaLucy, 2024-06-09:
finally got it that far, that at least opensbi boots by mapping it to different address space so it would fit into the 8 MiB. But now linux dont load and i have no idea how to debug it...
any suggestions?

```
[LITEX-TERM] Received firmware download request from the device.
[LITEX-TERM] Uploading images/Image to 0x40000000 (7531468 bytes)...
[LITEX-TERM] Upload complete (8.3KB/s).
[LITEX-TERM] Uploading images/rv32.dtb to 0x407f0000 (2765 bytes)...
[LITEX-TERM] Upload complete (8.3KB/s).
[LITEX-TERM] Uploading images/opensbi.bin to 0x40740000 (132040 bytes)...
[LITEX-TERM] Upload complete (8.3KB/s).
[LITEX-TERM] Booting the device.
[LITEX-TERM] Done.
Executing booted program at 0x40740000

--============= Liftoff! ===============--

OpenSBI v0.8-703-g84c6dc1
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : LiteX / VexRiscv-SMP
Platform Features         : medeleg
Platform HART Count       : 8
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 100000000Hz
Platform Console Device   : litex_uart
Platform HSM Device       : ---
Platform PMU Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Platform Suspend Device   : ---
Platform CPPC Device      : ---
Firmware Base             : 0x40740000
Firmware Size             : 248 KB
Firmware RW Offset        : 0x20000
Firmware RW Size          : 120 KB
Firmware Heap Offset      : 0x32000
Firmware Heap Size        : 48 KB (total), 3 KB (reserved), 8 KB (used), 36 KB (free)
Firmware Scratch Size     : 4096 B (total), 452 B (used), 3644 B (free)
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*,1*,2*,3*,4*,5*,6*,7*
Domain0 Region00          : 0xf0018000-0xf001bfff M: (I,R,W) S/U: ()
Domain0 Region01          : 0xf0010000-0xf0017fff M: (I,R,W) S/U: ()
Domain0 Region02          : 0x40740000-0x4075ffff M: (R,X) S/U: ()
Domain0 Region03          : 0x40760000-0x4077ffff M: (R,W) S/U: ()
Domain0 Region04          : 0x00000000-0xffffffff M: (R,W,X) S/U: (R,W,X)
Domain0 Next Address      : 0x40000000
Domain0 Next Arg1         : 0x407f0000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes
Domain0 SysSuspend        : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART Priv Version    : v1.10
Boot HART Base ISA        : rv32ima
Boot HART ISA Extensions  : zicntr
Boot HART PMP Count       : 0
Boot HART PMP Granularity : 0
Boot HART PMP Address Bits: 0
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x00000222
Boot HART MEDELEG         : 0x0000b109
```

and here is my address table:
```
	start		ende	size		needed	name

0x40000000 - 0x4072ffff	0x730000	72ebcc	kernel
0x40730000 - 0x4073ffff	0x10000				(free)
0x40740000 - 0x4076ffff	0x30000		203c8	opensbi		
0x40770000 - 0x407effff	0x80000				(free)
0x407f0000 - 0x407fffff	0x10000		?		device tree
```

Dolu1990, 2024-06-10:
Running it in a simulation can help figureing out where the CPU is going. For early boot issues it can help.

You may have an issue in your dts / missing linux drivers aswell

LaneaLucy, 2024-06-10:
it looks like its a opensbi problem. now got a tang primer 20k and with opensbi from here, it boots into linux, but with self compiled version it doesn't boot into Linux

LaneaLucy, 2024-06-12:
I don't know how, but it works now with self compiled everything

closed 2024-06-12 by LaneaLucy