litex-hub/linux-on-litex-vexriscv #394: ERROR:SoC:LOCALIP1 Constant already declared

jjagielo, 2024-06-13:
I am running:
```bash
$ ./make.py --board=arty_a7 --cpu-count=1 --build
```
and I am getting the following error (Same as the title):
```bash
ERROR:SoC:LOCALIP1 Constant already declared.
```
Is there something I need to manually change in my environment to remove this already declared constant?

jjagielo, 2024-06-13:
I am also now getting the following error, after following this thread: https://gist.github.com/ibndias/6f44d1d55251324f5b873436ad290ab3?permalink_comment_id=4938520#gistcomment-4938520

```bash
jason@jason-VirtualBox:~/Desktop/LiteX/linux-on-litex-vexriscv$ ./make.py --board=arty_a7 --cpu-count=1 --build
Traceback (most recent call last):
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 844, in <module>
    main()
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 733, in main
    board = supported_boards[board_name]()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 76, in __init__
    from litex_boards.targets import arty
ImportError: cannot import name 'arty' from 'litex_boards.targets' (/usr/local/lib/python3.12/dist-packages/litex_boards-2024.4-py3.12.egg/litex_boards/targets/__init__.py)
```

trabucayre, 2024-06-14:
Fixed by [6f65868](https://github.com/litex-hub/linux-on-litex-vexriscv/commit/6f65868843737567f9a9128d43b9d1cd36dfd1ae)
Since some time soc/add_ethernet add localip/remoteip constants automatically when parameter are not None (here only local_ip is provided so no more need to do that in soc_linux.py)

jjagielo, 2024-06-17:
Did a pull from the master branch and everything is resolved, thanks!

closed 2024-06-17 by jjagielo