litex-hub/linux-on-litex-vexriscv #395: ImportError: cannot import name '{board name}' from 'litex_boards.targets'

jjagielo, 2024-06-14:
I am getting the following error when trying to build the FPGA bitstrem:

```bash
jason@jason-VirtualBox:~/Desktop/LiteX/linux-on-litex-vexriscv$ ./make.py --board=arty --cpu-count=1 --build
Traceback (most recent call last):
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 844, in <module>
    main()
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 733, in main
    board = supported_boards[board_name]()
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jason/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 76, in __init__
    from litex_boards.targets import arty
ImportError: cannot import name 'arty' from 'litex_boards.targets' (/home/jason/Desktop/LiteX/litex-boards/litex_boards/targets/__init__.py)
```

I have also tried to change 'arty' to other valid board arguments, but it keeps throwing this exact error.

trabucayre, 2024-06-14:
According to [make.py](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/make.py#L76) your local repository is outdated.
Please update it.

jjagielo, 2024-06-14:
I ran _./litex_setup.py --update_ and that did not change anything from before. Is there a specific way to update this, as in which repositories need to be updated? The one featuring _./make.py_ or the _litex-build_ repository?

trabucayre, 2024-06-14:
No `litex_setup.py` is used to install the framework and `make.py` to simplify build.

I assume you have used `git clone xxx` to have a local copy of this repository, no?
So you have to use `git pull` to update this copy.

jjagielo, 2024-06-14:
Gotcha, just in _./linux-on-litex-vexriscv_ is the only place where I need to pull the most recent version?

trabucayre, 2024-06-14:
You have already updated frameworks with `litex_setup.py`: so yes.

jjagielo, 2024-06-14:
Perfect, all working now. Thank you for your help!

closed 2024-06-14 by jjagielo

trabucayre, 2024-06-14:
Great!