litex-hub/linux-on-litex-vexriscv #396: Support for 1bpp framebuffer

machdyne, 2024-06-19:
I'm interested in getting a 1bpp black and white framebuffer working in order to improve video performance where color isn't needed.

My understanding is that this would require some changes to LiteX and simplefb in the Linux kernel:

https://stackoverflow.com/questions/67397306/monochromatic-1bpp-simple-framebuffer-simplefb-c-format-depth

Am I right in thinking that this could significantly speed up the text console and possibly X11 also? Does anyone else think this is a good idea?

enjoy-digital, 2024-07-09:
Hi @machdyne,

this could be interesting yes. We would first need to add 1bpp support to LiteX's framebuffer and test it. Then we could see to add Linux support and enable it for the project.

machdyne, 2024-07-11:
Thanks @enjoy-digital,

I was able to get this working, and the console appears to be drawing faster with 1bpp mono, but I wonder if it could be improved further.

https://github.com/machdyne/litex-linux/tree/litex-monochrome

I had to manually update the DTS file:

```
            framebuffer0: framebuffer@40c00000 {
                compatible = "simple-framebuffer";
                reg = <0x40c00000 0x9600>;
                width = <640>;
                height = <480>;
                stride = <80>;
                format = "mono1";
            };
```