litex-hub/linux-on-litex-vexriscv #397: Utilizing the PL to DDR interface of Zynq to define `main_ram`

jahagirdar, 2024-06-23:
I see a few closed issues on this topic indicating that DDR in Zynq (Pynq-Z2) cannot be used because it is not accessible from PL.
When I look at the Zynq7 block design I see 4 AXI 32/64b ports from PL to Memory Interface. Is there some way in which it can be used to extend main_ram?
[image]

Dolu1990, 2024-06-24:
Hi,

I don't know about the details why vex isn't ok on Zynq, but we recently got VexiiRiscv to work on it : 
https://github.com/enjoy-digital/litex/pull/1989#issuecomment-2184223779

Would that work for you ? (using vexii instead of vex)

jahagirdar, 2024-06-24:
Thanks @Dolu1990 for the Vexii ref.
I have recently started looking at the Vex* and litex ecosystem via the CFU-Playground project https://github.com/google/CFU-Playground . 
While the project looks to be another abandonware by google,  I am currently trying to go through the CFU  tutorial and was trying to get it working with the board available with me (pynq-z2)
I currently do no have sufficient knowledge regarding Vex and liteX to change the internals of the CFU Playground project and make it point to Vexii :( Any pointers for an experienced RTL-FPGA person to learning LiteX? I have worked with RocketChip and Bluespec cores )

Dolu1990, 2024-06-24:
Hoo then better you stay on vex itself for now, as vexii doesn't provide CFU yet.
I don't know enough the litex things (related to the PS / interconnect)

enjoy-digital, 2024-07-09:
@jahagirdar: Using RAM of the Zynq from LiteX design can indeed be useful in some  cases (like yours) but it's not currently officially supported by the framework. There have been some experiment as pointed by @Dolu1990 and we'll probably integrate support in the future.

For now, to learn follow CFU playground and learn about Vex/Vexii and LiteX, I would recommend switching to a board already supported by the project (not necessarily expensive like the ColorLight or QMTech ones).