litex-hub/linux-on-litex-vexriscv #399: rror: unable to open ftdi device with description '*', serial '*' at bus location '*'

jjagielo, 2024-06-25:
I am trying to run

```bash
./make.py --board=arty --cpu-count=1 --load
```

and I am currently getting an error below, relating to my USB connection.

```bash
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'

Traceback (most recent call last):
  File "/home/cdluser/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 988, in <module>
    main()
  File "/home/cdluser/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 977, in main
    board.load(filename=builder.get_bitstream_filename(mode="sram"))
  File "/home/cdluser/Desktop/LiteX/linux-on-litex-vexriscv/./make.py", line 33, in load
    prog.load_bitstream(filename)
  File "/home/cdluser/Desktop/LiteX/litex/litex/build/openocd.py", line 28, in load_bitstream
    self.call(["openocd", "-f", config, "-c", script])
  File "/home/cdluser/Desktop/LiteX/litex/litex/build/generic_programmer.py", line 101, in call
    raise OSError(msg)
OSError: Error occured during OpenOCD's call, please check:
- OpenOCD installation.
- Access permissions.
- Hardware and cable.
- Bitstream presence.
```

Is there a specific way to pinpoint what could be causing, as i am able to see that my FPGA is plugged in properly:

```bash
$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 1532:028a Razer USA, Ltd Razer Blade
Bus 001 Device 003: ID 03f0:1198 HP, Inc HID-compliant mouse
Bus 001 Device 006: ID 221a:0100 ZTEX GmbH FPGA Boards                          # Connected Properly
Bus 001 Device 002: ID 13d3:5279 IMC Networks Integrated RGB Camera
Bus 001 Device 005: ID 8087:0033 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

trabucayre, 2024-06-25:
Your target is an arty. This one has an FTDI with official VID/PID (0403:6010), but there is no FTDI in your `lspci` dump.
Are you sure to have an arty board?

jjagielo, 2024-06-25:
I have an [Artix 7](https://www.ztex.de/usb-fpga-2/usb-fpga-2.13.e.html), but I redid it with the target as arty_a7 and got the same result. I followed the installation as instructed.

jjagielo, 2024-06-25:
This is what I get from `lpsci`

```bash
$ lspci
00:00.0 Host bridge: Intel Corporation 12th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:01.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x16 Controller #1 (rev 02)
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-P Integrated Graphics Controller (rev 0c)
00:04.0 Signal processing controller: Intel Corporation Alder Lake Innovation Platform Framework Processor Participant (rev 02)
00:06.0 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #0 (rev 02)
00:06.2 PCI bridge: Intel Corporation 12th Gen Core Processor PCI Express x4 Controller #2 (rev 02)
00:08.0 System peripheral: Intel Corporation 12th Gen Core Processor Gaussian & Neural Accelerator (rev 02)
00:0a.0 Signal processing controller: Intel Corporation Platform Monitoring Technology (rev 01)
00:14.0 USB controller: Intel Corporation Alder Lake PCH USB 3.2 xHCI Host Controller (rev 01)
00:14.2 RAM memory: Intel Corporation Alder Lake PCH Shared SRAM (rev 01)
00:14.3 Network controller: Intel Corporation Alder Lake-P PCH CNVi WiFi (rev 01)
00:15.0 Serial bus controller: Intel Corporation Alder Lake PCH Serial IO I2C Controller #0 (rev 01)
00:16.0 Communication controller: Intel Corporation Alder Lake PCH HECI Controller (rev 01)
00:1c.0 PCI bridge: Intel Corporation Device 51bc (rev 01)
00:1d.0 PCI bridge: Intel Corporation Device 51b0 (rev 01)
00:1f.0 ISA bridge: Intel Corporation Alder Lake PCH eSPI Controller (rev 01)
00:1f.3 Multimedia audio controller: Intel Corporation Alder Lake PCH-P High Definition Audio Controller (rev 01)
00:1f.4 SMBus: Intel Corporation Alder Lake PCH-P SMBus Host Controller (rev 01)
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-P PCH SPI Controller (rev 01)
01:00.0 VGA compatible controller: NVIDIA Corporation GA106M [GeForce RTX 3060 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation Device 228e (rev a1)
03:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9A1/PM9A3/980PRO
04:00.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
05:00.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
05:01.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
05:02.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
05:03.0 PCI bridge: Intel Corporation Device 1133 (rev 02)
06:00.0 USB controller: Intel Corporation Device 1134
3a:00.0 USB controller: Intel Corporation Device 1135
6e:00.0 SD Host controller: O2 Micro, Inc. SD/MMC Card Reader Controller (rev 01)
```
In the link above, I know for sure that I have an Artix-7... is there something else I need to specify?

trabucayre, 2024-06-26:
Ok: with the board's link it's more clear ;-)
It's not an arty but another artix7 based board (maybe compatible -> you have to check/compare pinout, RAM, etc...).
With the arty you have an FTDI to program/load bitstream and to have an serial interface, here its through an FX2 and/or an external JTAG interface: this why openocd is unable to find an FTDI device.

trabucayre, 2024-06-27:
One thing not clear to me: why using [digilent arty a7](https://digilent.com/reference/programmable-logic/arty-a7/start)?
Your board is supported by litex-boards, you have to add a new entry/config in make.py.
About JTAG interface: the best way is to connect an external USB<->JTAG to CON5.

jjagielo, 2024-06-27:
So I have the external USB <-> JTAG to CON5 connected. I'm not sure why the error from OpenOCD is popping up. Is there something else wrong?

[image]

trabucayre, 2024-06-27:
Jack is for power supply
mini USB connector for FX2
Could you tell me name of the external USB<-> you have used?
If you persist to use arty target, errors may be:
-  a different FTDI than the one on this board
- access right: your user is not in the FTDI's group owner

jjagielo, 2024-06-27:
Are you referencing the cord I am using, or the port I am using?

trabucayre, 2024-06-27:
Port. Both on this picture aren't relative to the programming with and FTDI external device.
You have to populate CON5 (near RAM chip) and connect your external probe here.

trabucayre, 2024-06-27:
Once again: CON2 is connected to an FX2 device, with a specific protocol. It's not an FTDI device, it's not compatible with an FTDI device and consequently openocd will not be able to do something with this device.
Yes I assume it's possible to use it with a tool provided by ztex. But until to you stop to confuse between ztex board and digilent board you will not be able to load/program your board.
And since this board it's not an arty the bitstream generated will not works with your board. I haven't checked the schematic but there is a great chance the pinout is different than digilent arty board (both board are based on ARTIX but it's two differents boards) (I have pointed correct platform/target).

enjoy-digital, 2024-07-09:
Hi @jjagielo,

please first read https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support. Here it seems you are confusing Arty (the Digilent Arty) with Artix-7 (the Xilinx FPGA). To run the project on your board, you first need to add support for your board in the project, which is not currently the case. 

However, it seems that the support is present in LiteX-Boards, so running the project on it would be possible. As a first step, you'll have to get test the LiteX-Boards target on your hardware: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/ztex213.py

When you get it working with default settings and are able to load your FPGA, adding the support for the project will consist at listing it on `board.py` with the right configuration.

closed 2024-07-09 by enjoy-digital