litex-hub/linux-on-litex-vexriscv #403: Sipeed Tang Primer 20k doesn't seem to fit

pepijndevos, 2024-08-23:
When I run `./make.py --board=sipeed_tang_primer_20k --cpu-count=32 --build` with Gowin IDE 1.9.9 or 1.9.8 it fails to place the design because it has too few of everything by an order of magnitude:

```
ERROR (RP0002) : The number(64) of DSP in the design exceeds the resource limit(12) of current device. And DSP_STYLE maybe the useful user assignment to change the inference result
ERROR (RP0002) : The number(932) of BSRAM in the design exceeds the resource limit(46) of current device. And RAM_STYLE maybe the useful user assignment to change the inference result
ERROR (RP0001) : The number(64) of ALU54D in the design exceeds the resource limit(24) of current device(GW2A-LV18PG256C8/I7)
ERROR (RP0001) : The number(128) of MULT18X18 in the design exceeds the resource limit(48) of current device(GW2A-LV18PG256C8/I7)
ERROR (RP0001) : The number(96483) of DFF(DL) in the design exceeds the resource limit(16173) of current device(GW2A-LV18PG256C8/I7)
```

Supposedly this board worked at some point, so this is somewhat puzzling. Were there any recent configuration changes that significantly increased resources? Which version of Gowin IDE is known to work?

pepijndevos, 2024-08-23:
Wait... that's FPGA cores, not host cores isn't it...

closed 2024-08-23 by pepijndevos

Dolu1990, 2024-08-27:
<3<3<3 no worries ^^