litex-hub/linux-on-litex-vexriscv #404: serial boot from BIOS on SoClinux stuck.

SeanGan233, 2024-09-18:
hi everyone! i am testing demo on tang primer 20k board. Something confuses me.
when i use `**python3 -m** litex_boards.targets.sipeed_tang_primer_20k --build --load` to build, and `litex_term /dev/ttyUSB1 --kernel=demo.bin` to do serial boot , i can boot demo successfully.
but when i use `**./make.py** --board=sipeed_tang_primer_20k --cpu-count=1 --build --load` to build, and `litex_term /dev/ttyUSB1 --kernel=demo.bin` to do serial boot, it stucks at `--============= Liftoff! ===============--`
Does the SoClinux have problems do serial boot? Can I do serial boot before booting linux?
any help appreciated.

SeanGan233, 2024-09-18:
I print the memory space(main sram 0x40000000) after serial booting, and found that the value was exactly the same in both cases. Apparently the program has been written to the appropriate memory space, but somehow it does not boot from the space successfully.

jeremy-heath, 2024-09-18:
I had this problem when I had a kernel larger than the memory region. Make sure that your kernel is not larger than the memory region (typically 8MB).

closed 2024-09-20 by SeanGan233