litex-hub/linux-on-litex-vexriscv #405: LiteX simulation has no VexRiscv pregenerated configuration

GMXX, 2024-09-25:
According to the documentation, some regular VexRiscv-smp configurations are already pre-generated, but when running `sim.py`, it starts by creating an SoC instead of booting:

```
INFO:SoC:Initializing ROM rom with contents (Size: 0x64f8).
INFO:SoC:Auto-Resizing ROM rom from 0x10000 to 0x64f8.
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-09-24 19:10:15)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 100.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
```

The README instructions worked as expected some months ago.

Dolu1990, 2024-09-25:
Fixed :)

GMXX, 2024-09-25:
Hi, @Dolu1990.

I've just tested it again on a clean, from-scratch install, and the problem persists, i.e., it is still trying to create the SoC rather than using the pre-generated. I double-checked, and both the commits were added, as expected.

Can you take a further look, please?

Dolu1990, 2024-09-27:
When you run sim.py, you still get : 

```
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
```

And nothing else ?
No idea were Litex is stuck / waiting ?

GMXX, 2024-09-28:
Yes, I still get it. Here is the full output of `sim.py`: 

```
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2024-09-26 13:34:02)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : SIM.
INFO:SoC:System clock: 100.000MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Controller ctrl added.
INFO:SoC:CPU vexriscv_smp added.
INFO:SoC:CPU vexriscv_smp adding IO Region 0 at 0x80000000 (Size: 0x80000000).
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode:  RW, Cached: False, Linker: False.
INFO:SoC:CPU vexriscv_smp overriding sram mapping from 0x01000000 to 0x10000000.
INFO:SoC:CPU vexriscv_smp setting reset address to 0x00000000.
INFO:SoC:CPU vexriscv_smp adding Bus Master(s).
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoC:CPU vexriscv_smp adding Interrupt(s).
INFO:SoCIRQHandler:noirq IRQ added at Location 0.
INFO:SoC:CPU vexriscv_smp adding SoC components.
INFO:SoCCSRHandler:uart CSR added at Location 2.
INFO:SoCCSRHandler:timer0 CSR added at Location 3.
INFO:SoCBusHandler:opensbi Region added at Origin: 0x40f00000, Size: 0x00080000, Mode:  RW, Cached:  True, Linker:  True.
INFO:SoCBusHandler:plic Region added at Origin: 0xf0c00000, Size: 0x00400000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:plic added as Bus Slave.
INFO:SoCBusHandler:clint Region added at Origin: 0xf0010000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:clint added as Bus Slave.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00010000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00010000, Mode:  RX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 1.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 2.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x04000000, Mode: RWX, Cached:  True, Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:CSR Bridge csr added.
INFO:SoCBusHandler:csr Region added at Origin: 0xf0000000, Size: 0x00010000, Mode:  RW, Cached: False, Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:csr added as CSR Master.
INFO:SoCBusHandler:Interconnect: InterconnectShared (1 <-> 6).
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCCSRHandler:sdram CSR allocated at Location 1.
INFO:SoCCSRHandler:supervisor CSR allocated at Location 4.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Finalized SoC:
[... 171 lines cut ...]
│    │    └─── csrstatus_1* (CSRStatus)
│    │    └─── csrstatus_2* (CSRStatus)
│    │    └─── csrstatus_3* (CSRStatus)
│    │    └─── csrstorage_0* (CSRStorage)
│    │    └─── csrstatus_4* (CSRStatus)
│    │    └─── csrstatus_5* (CSRStatus)
└─── csr_interconnect (InterconnectShared)
* : Generated name.
[]: BlackBox.

INFO:SoC:--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/content/linux-on-litex-vexriscv/./sim.py", line 179, in <module>
    main()
  File "/content/linux-on-litex-vexriscv/./sim.py", line 173, in main
    builder.build(sim_config=sim_config, run=run, **verilator_build_kwargs)
  File "/content/litex/litex/soc/integration/builder.py", line 393, in build
    self._check_meson()
  File "/content/litex/litex/soc/integration/builder.py", line 311, in _check_meson
    raise OSError(msg)
OSError: Unable to find valid Meson build system, please install it with:
- pip3 install meson.
```

It is complaining about meson, but once I install it, it will complain about everything else needed to build the SoC (ninja, etc.) until it can finally fully build it instead of just booting the pre-generated.

Dolu1990, 2024-09-28:
> some regular VexRiscv-smp configurations are already pre-generated

Yes, the verilog of the VexRiscv soc part is pregenerated.

Still it needs to compile the software which will run on it.

Those other dependencies (ninja / meson) are requirements with no pregeneration possibilities.

GMXX, 2024-09-30:
It makes sense now. From my recollection of the last time I used it, I thought the simulation would start straight from the Linux boot phase. 

Thank you, @Dolu1990!

closed 2024-09-30 by GMXX