litex-hub/linux-on-litex-vexriscv #406: USB OHCI peripheral netlist generation issue

Basman74, 2024-09-25:
New Litex user (via Ubuntu under WSL2) here. Everything so far works great (!) with the exception of one item. 
I seem to be having an issue generating a custom netlist for the usb_ohci.py peripheral: sbt seems to throw an error ([screenshot](https://imgur.com/a/eagIsEi)) when I attempt to increase the number of usb host ports beyond one. Suspect this may have worked previously for others, so I am wondering if it could be a new issue or just my own personal hell. Thanks in advance! 🙂

Basman74, 2024-09-25:
fleasys@DESKTOP-903IAF4:~/pythondata-misc-usb_ohci/pythondata_misc_usb_ohci/verilog/ext/SpinalHDL$ tree -L 6
.
├── project
│   ├── build.properties
│   ├── project
│   │   ├── build.properties
│   │   └── target
│   │       ├── scala-2.12
│   │       │   └── sbt-1.0
│   │       │       ├── sync
│   │       │       └── update
│   │       └── streams
│   │           ├── _global
│   │           │   ├── _global
│   │           │   ├── csrConfiguration
│   │           │   ├── csrProject
│   │           │   ├── dependencyPositions
│   │           │   ├── ivyConfiguration
│   │           │   ├── ivySbt
│   │           │   ├── moduleSettings
│   │           │   ├── projectDescriptors
│   │           │   ├── scalaCompilerBridgeScope
│   │           │   └── update
│   │           ├── compile
│   │           │   ├── _global
│   │           │   ├── bspReporter
│   │           │   ├── compile
│   │           │   ├── compileIncremental
│   │           │   ├── copyResources
│   │           │   ├── dependencyClasspath
│   │           │   ├── exportedProducts
│   │           │   ├── externalDependencyClasspath
│   │           │   ├── incOptions
│   │           │   ├── internalDependencyClasspath
│   │           │   ├── managedClasspath
│   │           │   ├── scalacOptions
│   │           │   ├── unmanagedClasspath
│   │           │   └── unmanagedJars
│   │           └── runtime
│   │               ├── dependencyClasspath
│   │               ├── exportedProducts
│   │               ├── externalDependencyClasspath
│   │               ├── fullClasspath
│   │               ├── internalDependencyClasspath
│   │               ├── managedClasspath
│   │               ├── unmanagedClasspath
│   │               └── unmanagedJars
│   └── target
│       ├── global-logging
│       ├── scala-2.12
│       │   └── sbt-1.0
│       │       ├── sync
│       │       │   └── copy-resource
│       │       └── update
│       │           └── update_cache_2.12
│       ├── streams
│       │   ├── _global
│       │   │   ├── _global
│       │   │   │   └── _global
│       │   │   ├── csrConfiguration
│       │   │   │   └── _global
│       │   │   ├── csrProject
│       │   │   │   └── _global
│       │   │   ├── dependencyPositions
│       │   │   │   └── _global
│       │   │   ├── ivyConfiguration
│       │   │   │   └── _global
│       │   │   ├── ivySbt
│       │   │   │   └── _global
│       │   │   ├── moduleSettings
[... 166 lines cut ...]
    │       ├── externalDependencyClasspath
    │       │   └── _global
    │       │       └── streams
    │       ├── fullClasspathAsJars
    │       │   └── _global
    │       │       └── streams
    │       ├── internalDependencyAsJars
    │       │   └── _global
    │       │       └── streams
    │       ├── managedClasspath
    │       │   └── _global
    │       │       └── streams
    │       ├── unmanagedClasspath
    │       │   └── _global
    │       │       └── streams
    │       └── unmanagedJars
    │           └── _global
    │               └── streams
    └── task-temp-directory
        ├── sbt_26d09ed9
        ├── sbt_8a867423
        ├── sbt_e440489d
        └── sbt_fb9a3dee

244 directories, 13 files

Dolu1990, 2024-09-25:
Ahh no core or lib folder ?

 tree -L 2

├── lib
│   ├── src
│   └── target

Basman74, 2024-09-25:
Correct. lib folder does not exist.

Dolu1990, 2024-09-25:
i think the git clone of that SpinalHDL folder failed in some ways.

Basman74, 2024-09-25:
Looks like I cannot even generate a custom vexriscv-smp variant reliably :(

Basman74, 2024-09-25:
because that is what I am using right now

Dolu1990, 2024-09-25:
right, no known issue.
More generaly, your issue seems related to git clone / network
Maybe try to reinstall litex to force recloning everything, also, check that those SpinalHDL have their lib and core folders, else, there is no need to look futher :)

Basman74, 2024-09-25:
Understood heh, sorry for the verbose tree listing :)

Basman74, 2024-09-25:
I'll backup what I have done (I have covered a _lot_ of ground here in a few short days), and retry. I will report back on the outcome. Many thanks for your help with this :)

Basman74, 2024-09-26:
I have an update. After doing the following in my Ubuntu 22.04 WSL2 environment: 
1.) Performed fresh install of sbt 1.10.1, Java 17.0.12 and Scala 2.11.12 
2.) Applied the '--recursive' option to my git clone of the Litex repo, solving the problem of missing files in the project tree

I now get the following result whenever I try and build a vexriscv-smp SoC:

_**[info] Non-compiled module 'compiler-bridge_2.12' for Scala 2.12.13. Compiling...
Killed**_

Basman74, 2024-09-26:
This is the (verbose) output:

fleasys@DESKTOP-903IAF4:~/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv$ sudo sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --coherent-dma --cpu-count=1 --reset-vector=0 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=128 --aes-instruction=False --expose-time=False --out-of-order-decoder=True --privileged-debug=False --hardware-breakpoints=1 --wishbone-memory=True --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Cdma_Ood_Wm_Hb1 --netlist-directory=/home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4 --jtag-tap=False" -mem2000
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by sbt.TrapExit$ (file:/root/.sbt/boot/scala-2.12.13/org.scala-sbt/sbt/1.5.0/run_2.12-1.5.0.jar)
WARNING: Please consider reporting this to the maintainers of sbt.TrapExit$
WARNING: System::setSecurityManager will be removed in a future release
[info] welcome to sbt 1.5.0 (Ubuntu Java 17.0.12)
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to VexRiscv (in build file:/home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[info] compiling 69 Scala sources and 2 Java sources to /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/target/scala-2.12/classes ...
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/Phase.scala:2615:25: non-variable type argument spinal.core.SpinalEnum in type pattern spinal.core.SpinalEnumCraft[spinal.core.SpinalEnum] is unchecked since it is eliminated by erasure
[warn]                 case d: SpinalEnumCraft[SpinalEnum] => d.init(d.spinalEnum.elements(0))
[warn]                         ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/AFix.scala:238:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match {
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/Data.scala:367:29: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, outWithNull
[warn]   def dirString(): String = dir match {
[warn]                             ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/Data.scala:541:16: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, outWithNull
[warn]           that.dir match {
[warn]                ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/SInt.scala:390:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match{
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/UInt.scala:269:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match{
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/Phase.scala:2501:12: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, inout, outWithNull
[warn]         io.dir match {
[warn]            ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/PhaseVerilog.scala:419:23: match may not be exhaustive.
[warn] It would fail on the following input: Nil
[warn]         val newName = IFlist match {
[warn]                       ^
Killed
fleasys@DESKTOP-903IAF4:~/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv$

Basman74, 2024-09-26:
```
fleasys@DESKTOP-903IAF4:~/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv$ sudo sbt "runMain vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --coherent-dma --cpu-count=1 --reset-vector=0 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=128 --aes-instruction=False --expose-time=False --out-of-order-decoder=True --privileged-debug=False --hardware-breakpoints=1 --wishbone-memory=True --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Cdma_Ood_Wm_Hb1 --netlist-directory=/home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog --dtlb-size=4 --itlb-size=4 --jtag-tap=False" -mem2000
WARNING: A terminally deprecated method in java.lang.System has been called
WARNING: System::setSecurityManager has been called by sbt.TrapExit$ (file:/root/.sbt/boot/scala-2.12.13/org.scala-sbt/sbt/1.5.0/run_2.12-1.5.0.jar)
WARNING: Please consider reporting this to the maintainers of sbt.TrapExit$
WARNING: System::setSecurityManager will be removed in a future release
[info] welcome to sbt 1.5.0 (Ubuntu Java 17.0.12)
[info] loading settings for project vexriscv-build from plugins.sbt ...
[info] loading project definition from /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] loading settings for project root from build.sbt ...
[info] loading settings for project spinalhdl-build from plugin.sbt ...
[info] loading project definition from /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/project
[info] loading settings for project all from build.sbt ...
[info] set current project to VexRiscv (in build file:/home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[info] compiling 69 Scala sources and 2 Java sources to /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/target/scala-2.12/classes ...
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/Phase.scala:2615:25: non-variable type argument spinal.core.SpinalEnum in type pattern spinal.core.SpinalEnumCraft[spinal.core.SpinalEnum] is unchecked since it is eliminated by erasure
[warn]                 case d: SpinalEnumCraft[SpinalEnum] => d.init(d.spinalEnum.elements(0))
[warn]                         ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/AFix.scala:238:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match {
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/Data.scala:367:29: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, outWithNull
[warn]   def dirString(): String = dir match {
[warn]                             ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/Data.scala:541:16: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, outWithNull
[warn]           that.dir match {
[warn]                ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/SInt.scala:390:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match{
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/UInt.scala:269:5: match may not be exhaustive.
[warn] It would fail on the following input: SCRAP
[warn]     roundType match{
[warn]     ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/Phase.scala:2501:12: match may not be exhaustive.
[warn] It would fail on the following inputs: inWithNull, inout, outWithNull
[warn]         io.dir match {
[warn]            ^
[warn] /home/fleasys/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/core/src/main/scala/spinal/core/internals/PhaseVerilog.scala:419:23: match may not be exhaustive.
[warn] It would fail on the following input: Nil
[warn]         val newName = IFlist match {
[warn]                       ^
Killed
fleasys@DESKTOP-903IAF4:~/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv$
/
```

Basman74, 2024-09-26:
That is as far as I get now

Basman74, 2024-09-26:
Never mind... bumping up the WSL2 available memory to 8GB did the trick! :)

Basman74, 2024-09-26:
The issue is now solved. You may now close this thread. :

Thanks for everything.. Litex is awesome!!! 👍 👍 👍 😀

closed 2024-09-26 by Basman74