litex-hub/linux-on-litex-vexriscv #409: Any luck generating a compiler on the target Linux image?

Rajnesh28, 2024-11-18:
Hi there!

I'd like to know if the folks behind this project have had any luck creating a Linux Image with a compiler on it. It looks like Buildroot doesn't support adding a compiler on the target, and suggests using other Linux distributions which would be much larger. There are a few "hacky" solutions out there involving packages and whatnot, but thought I'd reach out while I get my hands dirty on those.

If you have had luck, may I please know how you set up the configuration to make it, and/or have the image?

Cheers,

RJ

Dolu1990, 2024-11-21:
Hi,

Hmm a long time ago, i got gcc on buildroot, but it was a paine, and i do not have any trace remaining XD
it was compiled out of the buildroot package system, manualy.

Rajnesh28, 2024-12-13:
I figured out one way which was just editing the rootfs overlay, but until I setup a faster way to transfer files other than serial to LiteX bios on my FPGA then this is not worth it LOL

closed 2024-12-13 by Rajnesh28