litex-hub/linux-on-litex-vexriscv #411: Issues synthesizing bitstream

No-0n3, 2024-12-09:
Hi,

Been trying to synthesize a bitstream for ulx3s (85K) with "./make --board=ulx3s --device=LFE5U-85F --build" using the recompiled linux images.

I run into an error that I run out of "TRELLIS_SLICE" (180% approx. According to console) and it can't fulfill the syntezisis can't be fulfilled as resources aren't enough. This is strange as it should be for this board.

Tried to restrict the cpu-count but still the same issue that it uses more resources it seems.

If anybody can help.

Dolu1990, 2024-12-09:
Hi,

This is realy weird, because when this kind of LUT explosion comes, it is because of some memories not being infered as memory blocks.
That comes with the symptom of which flipflop utilisation. 

But in your case, it isn't the case XD : 
Info: Total DFFs: 14072/83640 16%

So, i realy have no idea what can have happened there.
Info: Total LUT4s: 150617/83640 180%  realy look crazy.

Could it be a ROM ? the being inferred as block ram but in logic instead ?

No-0n3, 2024-12-09:
After some debugging of the compilation, commented out the soc_capabilities in boards.py file for ulx3s (sdcard and framebuffer). Could build with sdcard but when building with framebuffer the issue comes up. Can boot into Linux without any problems if I build with the framebuffer soc capability is commented out. Is there any earlier bugs with the HDMI and video?

Dolu1990, 2024-12-10:
No idea.