litex-hub/linux-on-litex-vexriscv #416: Acorn - no output from console after load

wizath, 2025-03-21:
Hi,

I'm trying to run linux on `litex_acorn_baseboard_mini` and having some issues. Building with command 
`./make.py --board=acorn --cpu-count=2 --build --load`

Using ubuntu 20.04, Vivado 2018.3, followed the README steps

After flashing, the console output stays silent, no log even from bios. 

When I try to run the normal litex build, using
`python3 -m litex_boards.targets.sqrl_acorn` or `python3 -m litex_boards.targets.litex_acorn_baseboard_mini`, I can see it start and boot to the console
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2024 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 21 2025 02:06:24
 BIOS CRC passed (733bf3e6)

 LiteX git sha1: 88258bd34

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		wishbone 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128.0KiB
SRAM:		8.0KiB
L2:		8.0KiB
SDRAM:		1.0GiB 16-bit @ 800MT/s (CL-7 CWL-5)
MAIN-RAM:	1.0GiB

--========== Initialization ============--

Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11100000000000000000000000000000| delays: 01+-01
  m0, b02: |00001111111111111100000000000000| delays: 10+-06
  m0, b03: |00000000000000000000111111111111| delays: 25+-05
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 10+-06
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11100000000000000000000000000000| delays: 01+-01
  m1, b02: |00000111111111111110000000000000| delays: 11+-06
  m1, b03: |00000000000000000000011111111111| delays: 26+-05
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 11+-06
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 35.0MiB/s
   Read speed: 46.8MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--
```

Anyone have any idea why there is no output from linux build?

enjoy-digital, 2025-03-26:
Hi @wizath,

I'll look and fix if issues.

enjoy-digital, 2025-03-26:
@wizath: This should be good with https://github.com/litex-hub/linux-on-litex-vexriscv/commit/9493f31e8a9c12f866449e629d7497b8635d6c19.

I just tested `./make.py --board=acorn_baseboard_mini --cpu-count=2 --build --load `on hardware.

wizath, 2025-03-26:
It works now, thank you!
```
Welcome to Buildroot
buildroot login: root
                   __   _
                  / /  (_)__  __ ____ __
                 / /__/ / _ \/ // /\ \ /
                /____/_/_//_/\_,_//_\_\
                      / _ \/ _ \
   __   _ __      _  _\___/_//_/         ___  _
  / /  (_) /____ | |/_/__| | / /____ __ / _ \(_)__ _____  __
 / /__/ / __/ -_)>  </___/ |/ / -_) \ // , _/ (_-</ __/ |/ /
/____/_/\__/\__/_/|_|____|___/\__/_\_\/_/|_/_/___/\__/|___/
                  / __/  |/  / _ \
                 _\ \/ /|_/ / ___/
                /___/_/  /_/_/
  32-bit RISC-V Linux running on LiteX / VexRiscv-SMP.

login[75]: root login on 'console'
```

enjoy-digital, 2025-03-26:
Great, thanks for the feedback.