litex-hub/linux-on-litex-vexriscv #419: ZCU 104 - Cannot see nor configure LEDs on Buildroot

pcotret, 2025-04-25:
I've followed this tutorial for a ZCU104 board.
According to the [HOWTO](https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/HOWTO.md), I was hoping to play with the 4 LEDs available on the board.

However, I cannot see LEs in my Buildroot prompt:
```bash
root@buildroot:~# ls /sys/class/
bdi           graphics      mem           pps           tty
block         hwmon         misc          ptp           vc
devlink       i2c-adapter   mmc_host      pwm           vtconsole
drm           i2c-dev       mtd           spi_master
fpga_manager  input         net           spidev
```
LEDs are implemented in the Verilog design (I checked that to be sure).

LEDs are also visible in the device tree:
```
            leds: gpio@f0002800 {
                compatible = "litex,gpio";
                reg = <0xf0002800 0x4>;
                gpio-controller;
                #gpio-cells = <2>;
                litex,direction = "out";
                status = "disabled";
            };
```

Is the `status=disabled` important if I want to play with the LEDs?
How would I modify that?

pcotret, 2025-04-25:
https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/boards.py#L206

I've added the LED capability in the `boards.py` Python script but it doesn't change :(

trabucayre, 2025-04-25:
`status = "disabled"` means the driver will not be loaded. Could you try with `status = "okay"` ?

pcotret, 2025-04-28:
Hum, I just need to understand how to re-generate the software image once this `status="okay"` has been manually modified.

Rajnesh28, 2025-05-05:
@pcotret Not sure if the kernel supports overlays, but you can maybe use a device-tree overlay to override that status. 

Maybe it's changed since the last time I've used this repository, but I think there's a python script that generates the device-tree source based on some platform settings, maybe you could manually change that setting, recompile with `dtc` and then use that for your Linux boot?

pcotret, 2025-05-05:
> [@pcotret](https://github.com/pcotret) Not sure if the kernel supports overlays, but you can maybe use a device-tree overlay to override that status.
>
> Maybe it's changed since the last time I've [...]

I'll give it a try when I can