litex-hub/linux-on-litex-vexriscv #432: Linux hangs indefinitely in the init process

n1kl0s, 2025-09-08:
> I have the same problem with Linux getting stuck in the init process 

 _Originally posted by @n1kl0s in [#430](https://github.com/litex-hub/linux-on-litex-vexriscv/issues/430#issuecomment-3262193049)_

I get the following log. Even after hours, nothing is printed to the console. I have used the linux_on_litex_vexriscv_defconfig and the sim.py 

[    0.000000] Linux version 6.9.0 (mechelhoffn@niklas-Ubuntu) (riscv32-buildroot-linux-gnu-gcc.br_real (Buildroot 2025.05-249-gf050de7d9a-dirty) 13.4.0, GNU ld (GNU Binutils) 2.43.1) #1 SMP Sun Sep  7 16:16:14 CEST 2025
[    0.000000] Machine model: sim
[    0.000000] SBI specification v1.0 detected
[    0.000000] SBI implementation ID=0x1 Version=0x10003
[    0.000000] SBI TIME extension detected
[    0.000000] SBI IPI extension detected
[    0.000000] SBI RFENCE extension detected
[    0.000000] earlycon: liteuart0 at I/O port 0x0 (options '')
[    0.000000] Malformed early option 'console'
[    0.000000] earlycon: liteuart0 at MMIO 0xf0001000 (options '')
[    0.000000] printk: legacy bootconsole [liteuart0] enabled
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] mmode_resv1@40f00000 (0x40f00000--0x40f40000) overlaps with opensbi@40f00000 (0x40f00000--0x40f80000)
[    0.000000] OF: reserved mem: OVERLAP DETECTED!
[    0.000000] opensbi@40f00000 (0x40f00000--0x40f80000) overlaps with mmode_resv0@40f40000 (0x40f40000--0x40f60000)
[    0.000000] OF: reserved mem: 0x40f00000..0x40f3ffff (256 KiB) nomap non-reusable mmode_resv1@40f00000
[    0.000000] OF: reserved mem: 0x40f00000..0x40f7ffff (512 KiB) map non-reusable opensbi@40f00000
[    0.000000] OF: reserved mem: 0x40f40000..0x40f5ffff (128 KiB) nomap non-reusable mmode_resv0@40f40000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040efffff]
[    0.000000]   node   0: [mem 0x0000000040f00000-0x0000000040f5ffff]
[    0.000000]   node   0: [mem 0x0000000040f60000-0x0000000043ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x0000000043ffffff]
[    0.000000] SBI HSM extension detected
[    0.000000] riscv: base ISA extensions aim
[    0.000000] riscv: ELF capabilities aim
[    0.000000] percpu: Embedded 11 pages/cpu s22932 r0 d22124 u45056
[    0.000000] Kernel command line: console=liteuart earlycon=liteuart,0xf0001000 rootwait root=/dev/ram0
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] Memory: 47336K/65536K available (6676K kernel code, 579K rwdata, 1040K rodata, 258K init, 249K bss, 18200K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=32 to nr_cpu_ids=1.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 32 local interrupts mapped
[    0.000000] riscv: providing IPIs using SBI IPI extension
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x171024e7e0, max_idle_ns: 440795205315 ns
[    0.000024] sched_clock: 64 bits at 100MHz, resolution 10ns, wraps every 4398046511100ns
[    0.003145] Console: colour dummy device 80x25
[    0.004202] Calibrating delay loop (skipped), value calculated using timer frequency.. 200.00 BogoMIPS (lpj=1000000)
[    0.005150] pid_max: default: 32768 minimum: 301
[    0.009354] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.010144] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.051306] ASID allocator using 9 bits (512 entries)
[    0.056195] rcu: Hierarchical SRCU implementation.
[    0.056707] rcu: 	Max phase no-delay instances is 1000.
[    0.074010] smp: Bringing up secondary CPUs ...
[    0.074735] smp: Brought up 1 node, 1 CPU
[    0.085296] devtmpfs: initialized
[    0.128797] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.129715] futex hash table entries: 256 (order: 2, 16384 bytes, linear)
[    0.166552] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.178453] DMA: preallocated 128 KiB GFP_KERNEL pool for atomic allocations
[    0.311626] cpu0: Ratio of byte access time to unaligned word access is 0.00, unaligned accesses are slow
[    0.325644] platform soc: Fixed dependency cycle(s) with /soc/interrupt-controller@f0c00000
[    0.335187] platform soc: Fixed dependency cycle(s) with /soc/interrupt-controller@f0c00000
[    0.373547] pps_core: LinuxPPS API ver. 1 registered
[    0.374106] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.375192] PTP clock support registered
[    0.380792] FPGA manager framework
[    0.401011] clocksource: Switched to clocksource riscv_clocksource
[    0.728334] NET: Registered PF_INET protocol family
[    0.733728] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.746151] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.747088] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.747970] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.748973] TCP bind hash table entries: 1024 (order: 2, 16384 bytes, linear)
[    0.750021] TCP: Hash tables configured (established 1024 bind 1024)
[    0.752152] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.753133] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.776628] Unpacking initramfs...
[    0.802194] workingset: timestamp_bits=30 max_order=14 bucket_order=0
[    0.811832] io scheduler mq-deadline registered
[    0.812406] io scheduler kyber registered
[    0.813353] io scheduler bfq registered
[    0.836941] riscv-plic f0c00000.interrupt-controller: mapped 32 interrupts with 1 handlers for 2 contexts.
[    0.853708] LiteX SoC Controller driver initialized
[    4.671981] Freeing initrd memory: 8192K
[    5.909389] f0001000.serial: ttyLXU0 at MMIO 0x0 (irq = 12, base_baud = 0) is a liteuart
[    5.911288] printk: legacy console [liteuart0] enabled
[    5.911288] printk: legacy console [liteuart0] enabled
[    5.912218] printk: legacy bootconsole [liteuart0] disabled
[    5.912218] printk: legacy bootconsole [liteuart0] disabled
[    5.944747] i2c_dev: i2c /dev entries driver
[    5.975965] NET: Registered PF_INET6 protocol family
[    6.000501] Segment Routing with IPv6
[    6.003269] In-situ OAM (IOAM) with IPv6
[    6.005437] sit: IPv6, IPv4 and MPLS over IPv4 tunneling driver
[    6.023593] NET: Registered PF_PACKET protocol family
[    6.427516] clk: Disabling unused clocks
[    6.439697] Freeing unused kernel image (initmem) memory: 252K
[    6.440327] Kernel memory protection not selected by kernel config.
[    6.441789] Run /init as init process

n1kl0s, 2025-09-13:
Hi @Dolu1990, 

is there any news regarding the issue? Can you reproduce the issue?

Dolu1990, 2025-09-17:
commenting the uart interrupt of the DTS you used make it work : 
        liteuart0: serial@f0001000 {
            compatible = "litex,liteuart";
            reg = <0xf0001000 0x100>;
            //interrupts = <1>;
            status = "okay";
        };

Also, with interrupt i get that error : 

[    2.757233] irq 12: nobody cared (try booting with the "irqpoll" option)
[    2.762643] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 6.9.0 #2
[    2.768447] Hardware name: sim (DT)
[    2.771923] Call Trace:
[    2.774355] [<c000476c>] dump_backtrace+0x2c/0x3c
[    2.779040] [<c0670428>] show_stack+0x44/0x5c
[    2.783381] [<c067b414>] dump_stack_lvl+0x78/0xac
[    2.788069] [<c067b464>] dump_stack+0x1c/0x2c
[    2.792410] [<c0671958>] __report_bad_irq+0x5c/0xf4
[    2.797272] [<c0078d4c>] note_interrupt+0x308/0x358
[    2.802134] [<c00745c0>] handle_irq_event+0x108/0x118
[    2.807170] [<c007a3c8>] handle_fasteoi_irq+0xd8/0x350
[    2.812292] [<c0072d6c>] generic_handle_domain_irq+0x2c/0x48
[    2.817935] [<c032c268>] plic_handle_irq+0x9c/0x160
[    2.822797] [<c0072d6c>] generic_handle_domain_irq+0x2c/0x48
[    2.828441] [<c032bbfc>] riscv_intc_irq+0x34/0x80
[    2.833129] [<c067b4fc>] handle_riscv_irq+0x68/0xa8
[    2.837991] [<c06865d8>] call_on_irq_stack+0x44/0x58
[    2.842943] handlers:
[    2.845196] [<(ptrval)>] liteuart_interrupt
[    2.849363] Disabling IRQ #12
[    2.859235] Freeing unused kernel image (initmem) memory: 252K
[    2.864154] Kernel memory protection not selected by kernel config.
[    2.870125] Run /init as init process

Maybe a broken litex uart driver update, not sure.

maass-hamburg, 2025-09-19:
tx_pending is by the way never cleared in the linux driver, this could lead to tx_pending always being high and always interrupting as long as it is activated. If thats the problem, https://github.com/enjoy-digital/litex/pull/2319 could fit it.
In zephyr we had to do https://github.com/zephyrproject-rtos/zephyr/blob/9dfac5a66e2a84c41b3aa65707332c1e6c84452b/drivers/serial/uart_litex.c#L192-L198 to have it working with `EventSourceProcess` being used for the irq